DroidPlanner/Tower #1785: Tower Will not arm.

rogerhogan, 2016-08-08:
Everything is normal until It is time to hit "arm" and then I swipe the arrow to the right, at this time I get the words: sorry tower has stopped.
I am using:

Galaxy Tab A
Android Version 6.0.1
3DR Services v1.6.0 beta 2
Tower v3.2.1

roger
roger@railroadsusa.com

franknitty69, 2016-08-09:
confirmed.

galaxy tab s
android lollipop 5.0.2
3DR Services v1.6.0 beta 2
Tower v3.2.1

ne0fhyk, 2016-08-10:
@rogerhogan @franknitty69 The issue was caused because 3DR pushed a buggy version of 3DR Services (v1.6+).

It should go away if you revert to version 1.5.3 which can be found at https://github.com/dronekit/dronekit-android/releases/download/3DRServices-v1.5.3.4/3dr-services-prod-release.105304.apk

We're working to remove the dependency between Tower and 3DR Services to prevent such issues in the future.

closed 2016-08-10 by ne0fhyk

rogerhogan, 2016-08-10:
Thank You very much. I will try that today.

rogerhogan, 2016-08-10:
I did as instructed and every is back to working again. Thank You , Thank You for the help.