DroidPlanner/Tower #1794: Tower debugging possible ?

flo-vienna, 2016-08-20:
Is there any way of debugging if the Android Tower app tries to trigger the camera ?

It seems triggering via the Mission Planner app on PC seems to be very reliable but triggering via CH7 opt on the RC does (for unknown reasons) not seem to be 100% reliable and triggering via Tower / Tower beta app seems not to work while auto-mode and flying works (even with the footprints being shown on the tablet)

ne0fhyk, 2016-08-22:
@flo-vienna the upcoming version of Tower will allow to save/import the mission to/from a txt file as supported by mission planner.

So you can debug by saving the generated mission from Tower and loading it within mission planner to see what mission item is missing.

A beta for the new version should be available later this week.

ne0fhyk, 2016-08-22:
@flo-vienna in addition, have you tried adding a `DO_SET_SERVO` mission item when creating the mission in Tower as indicated [here](http://ardupilot.org/copter/docs/common-chdk-camera-control-tutorial.html#programming-camera-controls-in-mission-planner).

flo-vienna, 2016-08-22:
Hi 

I am using this cable http://www.mobilexcopter.com/files/Simple_multi_shutter_Pixhawk_settings.pdf

so it seems that using servo settings might not work as I am using the relay output that is "mirrored" by the cable and sent to the CHDK script.

How is the camera triggered by Tower and is there any way of tracking IF the camera is triggered ?

ne0fhyk, 2016-08-29:
Tower uses the do-set-cam-trigg-dist mission item (
http://ardupilot.org/copter/docs/mission-command-list.html#do-set-cam-trigg-dist)
to have the autopilot automatically trigger the camera shutter at regular
distance interval.

flo-vienna, 2016-08-30:
Thanks for your reply. 

Does this mean that I have to manually add this to my mission when using Structure Scanner or is it included implicitly ?  Do I need to add this when doing a survey of an area ? (because of the overlaps etc.)

flo-vienna, 2016-09-14:
Do I get the info in the latest beta version right that now the cam is triggered at the first waypoint which should pretty much solve my issues ?