DroidPlanner/Tower #1795: Add app credits page

ne0fhyk, 2016-08-21:
Add a page crediting the community developers.
Maybe allow users to contact the devs directly from that page (i.e: email address).

ne0fhyk, 2016-08-21:
@DroidPlanner/dev @DroidPlanner/lead-devs  @DroidPlanner/maintainers Thoughts on this?

kellyschrock, 2016-08-21:
Sounds good to me if it's good with everyone else.

squilter, 2016-08-22:
Sure. I'm not sure if email addresses would be helpful. How will a user know who to email regarding a particular feature?

kellyschrock, 2016-08-22:
That's a good question. What if, in the contributors list, there was a
little blurb about what they've worked on? It would be an extra maintenance
item.

ne0fhyk, 2016-08-22:
@kellyschrock @squilter The idea is to break down the credits page in several sub-sections (i.e: General, Map, Translations, Core Lib, etc) with each having the list of the devs that contributed to it.

With regard to the maintenance, each interested dev would submit a pr to include himself/herself within the credits page. As such the maintenance should be minimal and mostly consist in reviewing the dev's pr for inclusion.

kellyschrock, 2016-08-22:
Sounds like the best way to do it. I was going to suggest having the
contributing developers maintaining their section of this document to
minimize overhead, actually.