DroidPlanner/Tower #1796: Why I can not use channels 9 and 10?

cuchara, 2016-08-24:
Why I can not use channels 9 and 10?
I try to use as channels 9 and 10 of the transmitter with the APM, but does not work. why?

[image]

guiseco, 2016-08-24:
http://ardupilot.org/copter/docs/channel-7-and-8-options.html

cuchara, 2016-08-24:
thanks for your answer.

>  (http://ardupilot.org/copter/docs/channel-7-and-8-options.html)

I already had read before ... but still do not understand ... is the same as I did, not?

closed 2016-08-24 by ne0fhyk

cuchara, 2016-08-24:
@ne0fhykne0fhyk closed this 5 hours ago   way ?

guiseco, 2016-08-24:
APM cannot handle more than 8 channels. that´s it! Only PIXHAWK CAN !

============================================================
  ((
  )))
|''''''''''|-.    Sun Certified Java Programmer
|___|-'    Guilherme S. Blanco

============================================================

cuchara, 2016-08-25:
Hi thanks for your answer. 

 HobbyKing Daily - HKPilot 32 Controller  should work as  PIXHAWK , no?

guiseco, 2016-08-25:
Yes ! Is a Good FC !!!

cuchara, 2016-08-26:
So why I cannot handle more than 8 channels ?   : /