DroidPlanner/Tower #1798: Error:Error: Fragments should be static such that they can be re-instantiated by the system, and anonymous classes are not static [ValidFragment]

Jim-uav, 2016-08-26:


guiseco, 2016-08-26:
Thats lint error,  to disable aborting on errors, add:

lintOptions {
    abortOnError false
}

to build.gradle.

Jim-uav, 2016-08-26:
It was already in it, but it didn't work.
[image]

ne0fhyk, 2016-08-26:
@Jim-uav what's this issue about? There's no description given.

Jim-uav, 2016-08-27:
@ne0fhyk 
[image]

bitdust, 2016-09-06:
also have the same problem

Linjieqiang, 2016-09-06:
also have the same problem

guiseco, 2016-09-06:
Same problem to me.

bitdust, 2016-09-06:
i just solve it by add
"checkReleaseBuilds false"
in lintOptions

like this:
[image]

ne0fhyk, 2016-09-06:
Can you all try with the [`standalone`](https://github.com/DroidPlanner/Tower/tree/standalone) branch. I pushed an update that should also fix the issue.

bitdust, 2016-09-06:
@ne0fhyk 
cool~it works.

ne0fhyk, 2016-09-07:
Fixed by commit d0abadce17ed05415ea406b13036695d5149a217

closed 2016-09-07 by ne0fhyk