DroidPlanner/Tower #1801: Upgrade to safe version of libjpeg_turbo

ne0fhyk, 2016-09-06:
@guiseco The version of `libjpeg_turbo` that's bundled in the [`libuvccamera-release.aar`](https://github.com/DroidPlanner/Tower/blob/develop/Android/libs/libuvccamera-release.aar) file is reported unsafe by Google Play.
Can you please update to version 1.4.2 to fix the issue.

FYI, here's the unsafe notification email I received from Google Play:

> Hello Google Play Developer,
> 
> We detected that your app(s) listed at the end of this email are using an unsafe version of libjpeg_turbo library.
> 
> What’s happening
> 
> Beginning September 17, 2016, Google Play will block publishing of any new apps or updates that use pre 1.4.2 versions of libjpeg_turbo. Apps with vulnerabilities like this can expose users to risk of compromise and may be considered in violation of our Malicious Behavior policy.
> 
> Action required: Migrate your app(s) to use libjpeg-turbo v1.4.2 or higher as soon as possible and increment the version number of the upgraded APK.
>

ne0fhyk, 2016-09-06:
@guiseco let me know if you have any questions.

guiseco, 2016-09-06:
Yes , for sure !

ne0fhyk, 2016-09-07:
Fixed by pr #1803. Thanks @guiseco!

PR #1803 (2016-09-07): Standalone
Jpeg 1.5.0 Bug fix #1801
 Improve portuguese translation
closed 2016-09-07 by ne0fhyk