DroidPlanner/Tower #1810: Out of memory  - Editor - Structure Scanner

guiseco, 2016-09-15:
E/art     (13861): Throwing OutOfMemoryError "Failed to allocate a 6031872 byte allocation with 3291936 free bytes and 3MB until OOM"
E/Vold    (  426): Failed to find mounted volume for /storage/sdcard1/Android/data/org.droidplanner.android.beta/files/
W/ContextImpl(13861): Failed to ensure directory: /storage/sdcard1/Android/data/org.droidplanner.android.beta/files
E/AndroidRuntime(13861): FATAL EXCEPTION: pool-7-thread-1
E/AndroidRuntime(13861): Process: org.droidplanner.android.beta, PID: 13861
E/AndroidRuntime(13861): java.lang.OutOfMemoryError: Failed to allocate a 6031872 byte allocation with 3291936 free bytes and 3MB until OOM
E/AndroidRuntime(13861):        at java.util.ArrayList.addAll(ArrayList.java:198)
E/AndroidRuntime(13861):        at org.droidplanner.services.android.impl.core.mission.waypoints.StructureScannerImpl.packCircles(StructureScannerImpl.java:56)
E/AndroidRuntime(13861):        at org.droidplanner.services.android.impl.core.mission.waypoints.StructureScannerImpl.packMissionItem(StructureScannerImpl.java:40)
E/AndroidRuntime(13861):        at org.droidplanner.services.android.impl.core.mission.waypoints.StructureScannerImpl.getPath(StructureScannerImpl.java:89)
E/AndroidRuntime(13861):        at org.droidplanner.services.android.impl.utils.ProxyUtils.getProxyMissionItem(ProxyUtils.java:473)
E/AndroidRuntime(13861):        at org.droidplanner.services.android.impl.utils.CommonApiUtils.buildStructureScanner(CommonApiUtils.java:928)
E/AndroidRuntime(13861):        at org.droidplanner.services.android.impl.utils.CommonApiUtils.buildComplexMissionItem(CommonApiUtils.java:896)
E/AndroidRuntime(13861):        at org.droidplanner.services.android.impl.api.DroneApi.executeAction(DroneApi.java:399)
E/AndroidRuntime(13861):        at com.o3dr.android.client.Drone.performActionOnHandler(Drone.java:466)
E/AndroidRuntime(13861):        at com.o3dr.android.client.Drone.performActionOnDroneThread(Drone.java:459)
E/AndroidRuntime(13861):        at com.o3dr.android.client.Drone.performAction(Drone.java:455)
E/AndroidRuntime(13861):        at com.o3dr.android.client.apis.MissionApi.buildComplexMissionItem(MissionApi.java:235)
E/AndroidRuntime(13861):        at com.o3dr.android.client.apis.MissionApi.buildMissionItem(MissionApi.java:253)
E/AndroidRuntime(13861):        at com.o3dr.android.client.Drone$5.run(Drone.java:530)
E/AndroidRuntime(13861):        at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1112)
E/AndroidRuntime(13861):        at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:587)
E/AndroidRuntime(13861):        at java.lang.Thread.run(Thread.java:818)
I/octvm_klo( 5245): klo lock
I/WtEventController( 4082): Receive am_crash event for pid: 13861
W/ActivityManager( 1365):   Force finishing activity 1 org.droidplanner.android.beta/org.droidplanner.android.activities.EditorActivity

guiseco, 2016-09-15:
https://youtu.be/HC7BJVepW1E

ne0fhyk, 2016-09-21:
@guiseco Can you use Android Studio memory profiler to monitor the crash. It might be caused by a memory leak.

guiseco, 2016-09-21:
YEs ! i will do!

guiseco, 2016-09-22:
@ne0fhyk  I found the bug inside the **org.droidplanner.services.android.impl.core.mission.waypoints.StructureScannerImpl.java
** on method packCircles. When step height is ZERO we get stuck in the infinite loop.

[image]

[image]

guiseco, 2016-09-22:
[image]

ne0fhyk, 2016-09-22:
@guiseco nice catch. I'll push a pr to fix the issue.

ne0fhyk, 2016-09-22:
@guiseco I added a pull request to fix the issue: https://github.com/dronekit/dronekit-android/pull/446
Can you do a code review, thanks!

ne0fhyk, 2016-09-25:
Fixed by pr  dronekit/dronekit-android#446

closed 2016-09-25 by ne0fhyk