DroidPlanner/Tower #1811: Feature request: set CAM_TRIGG_DIST to 0 when staring an auto mission

ThorstenDP, 2016-09-21:
Hi all,

when setting up a mission Tower sets CAM_TRIGG_DIST to match the image overlap.
However, images are already taken when CAM_TRIGG_DIST is > 0 before the first waypoint is reached. Hence, it would be great if this could be avoided by setting CAM_TRIGG_DIST to 0 immediately after switching to Auto mode. I thought this was working before, but I might be wrong.
It would be great if this could be implemented/fixed.

Thanks and cheers,
Thorsten

PS: I have not tested the 4.0 beta so far

ne0fhyk, 2016-09-21:
I believe this is a duplicate of https://github.com/DroidPlanner/Tower/issues/1444 which is included in the 4.0 beta.

closed 2016-09-21 by ne0fhyk

ThorstenDP, 2016-09-21:
@ne0fhyk
Hi Fredia,

I forgot about #1444 but I am not sure if it is really the same. Btw, thanks a lot for looking into it!
So now (pre 4.0) it starts triggering after the first WP - or to be more precise it changes CAM_TRIGG_DIST to a reasonable value. What is needed in addition is to set CAM_TRIGG_DIST (CTD) to "0" when starting the mission.

We just had a flight where CTD as set to 7 in the parameter settings. At WP 1 the mission created with Tower changed it to 20m. So from takeoff to the first WP the camera tried to take lots of images - much too fast for the cameras capabilities. This can cause problems with georeferencing.
In the case you switch to RTL before the mission is finished or if there is some battery failsafe event the CTD set by Tower will remain in the parameter settings. So the next time you start a mission there will be this CDT - until the first WP.

Hence, setting CTD to 0 directly after switching to Auto mode would help a lot in avoiding unnecessary images as well as the hassle to set CDT to 0 before each flight manually.

If this is implemented already in 4.0 I'll be happy to test asap.
Cheers,
Thorsten