DroidPlanner/Tower #1812: Flight time estimation

TheAeroScout, 2016-09-21:
I’ve installed and used Tower Beta and am enjoying some of the extra features you have added. The one thing that I still can’t do, is that I am not getting an accurate flight time estimation when I create a mission in the editor.
 
I am using APM 3.1 in an original 3DR APM 2.6 board, the Tower beta app can see all my parameters and knows that I am using an Arduplane, however when it estimates the flight time it seems to think that my cruise speed is somewhere around 5m/s, when I actually fly at closer to 20m/s. I don't have an airspeed sensor fitted.

I'm not sure if this is a bug or if there is a parameter that Tower uses to estimate the flight time that I need to change.

You help would be greatly appreciated. Thanks in advance.

guiseco, 2016-09-23:
I think it's a bug !

ne0fhyk, 2016-09-25:
Fixed by pr #1815

closed 2016-09-25 by ne0fhyk