DroidPlanner/Tower #1813: Violent crash 15 meters high. After using Follow Me mode in Tower Beta Pixracer

chris669, 2016-09-22:
After trying successfully to activate the "Follow Me" mode with my Tower Beta interface (now using a Wifi interface via Telemetry port on Pixracer), I wanted to stop Guided Mode and selected the Stabilize Mode. Effect: All the motors switch off and my Tarot 650 falls at two meters from me. By luck, only the antenna support was broken and battery a littlebit crooked.
Any bad operation for me? How to leave the "Follow Me" mode? Where to report this bug?
Thanks,
Chris

guiseco, 2016-09-22:
Hi Cris! Post your issue here -> https://github.com/ArduPilot/ardupilot/issues
Don't forget to attach the log file

chris669, 2016-09-23:
Hi, thanks but I assume this crash is due to bad operation from me, as I didn't check the position of my throttle stick before changing the mode to Stabilize. Anyway, it could be good having a warning telling Attention, Throttle stick is not 50%. Could you tell me where to find the log? Thanks.