DroidPlanner/Tower #1823: Feature request: Integration of ublox M8P RTK

ThorstenDP, 2016-10-26:
Hi Fredia and all,

I just started a discussion oevr at discuss.ardupilot.org about the integration of ublox M8P RTK GPS:
http://discuss.ardupilot.org/t/integration-of-ublox-m8p-rtk-in-ardupilot/

@OXINARF mentioned there that Ardupilot already supports the GPS_RTCM_DATA MAVLink message. So my question is if you are planning support for this. I guess this would require parallel bluetooth and USB connections for telemetry and connecting the base.

Thanks and cheers,
Thorsten

OXINARF, 2016-10-26:
I'd just like to add that ArduPilot uses more bits in the flag present in that message. I hope that someone updates the message specification to make that clear.

ThorstenDP, 2017-03-27:
Hi all,
regarding a discussion about RTK GPS and dedicated telemetry links at http://discuss.ardupilot.org/t/m8p-rtk-setup-issues-with-hdop-baudrate-telemetry/16019
it seems GPS injection is the way to go...
Would be really great to see such a feature in tower one day.
Best regards,
Thorsten