DroidPlanner/Tower #1824: Rssi on top bar

janky69, 2016-10-28:
Hi, i have "-" in rssi value in top bar and the speach say "zero" dbm. Its problem only for me? Thank you!

guiseco, 2016-10-28:
same to me.

janky69, 2016-10-28:
I mean, it was previously value of mavlink radio signal and some bars in
phone signal strange style... But i am not sure...

trip5ter, 2016-11-01:
I have the Same issue, seems to be a problem with Tower 3 and tower 4 beta, when i use droid planner 2 RSSI information from the SiK radios is displayed correctly. 

Additionally, when I first installed Tower 3 from Google Play Store RSSI information was displayed, it was only after an update of Tower that the rssi was lost....

mariansoban, 2016-11-07:
I'm not quite sure, but - couldn't this be the issue?
Telemetry radios send their system/component like 
https://github.com/ArduPilot/SiK/blob/master/Firmware/radio/mavlink.c#L47-L49
_// use '3D' for 3DRadio
#define RADIO_SOURCE_SYSTEM '3'
#define RADIO_SOURCE_COMPONENT 'D'_

There was a change in Dronekit for Android lib, there are new conditions in **GenericMavLinkDrone.onMavLinkMessageReceived()** and **ArduPilot.onMavLinkMessageReceived()**:
`@Override
    public void onMavLinkMessageReceived(MAVLinkMessage message) {
        if (message.sysid != this.getSysid()) {
                // Reject Messages that are not for the system id
`

https://github.com/dronekit/dronekit-android/blob/develop/ClientLib/src/main/java/org/droidplanner/services/android/impl/core/drone/autopilot/apm/ArduPilot.java#L389
https://github.com/dronekit/dronekit-android/blob/develop/ClientLib/src/main/java/org/droidplanner/services/android/impl/core/drone/autopilot/generic/GenericMavLinkDrone.java#L578

guiseco, 2016-11-07:
I can not test in the simulator (dronekit-sitl). Apparently the simulator does not deliver that data.

billbonney, 2016-11-07:
I had the same issue in APM Planner. I can put in. PR to fix.

mariansoban, 2016-11-07:
Now I'm convinced, that the issue is the one I wrote before because of this:
https://github.com/dronekit/dronekit-android/blob/develop/ClientLib/src/main/java/org/droidplanner/services/android/impl/core/drone/autopilot/generic/GenericMavLinkDrone.java#L240-L243
`   public short getSysid() {`
`        return heartbeat.getSysid();`
In other words, component ID to compare is taken from heartbeat packet - this value should be '1', which is component ID of vehicle.
Telemetry radio sets component ID to '3' for radio status packet and this packet is later filtered in Tower on those lines in _onMavLinkMessageReceived()_ method.

guiseco, 2016-11-07:
> ClientLib/src/main/java/org/droidplanner/services/android/
> impl/core/drone/autopilot/generic/GenericMavLinkDrone.java#L240-L243

Nice catch !

guiseco, 2016-11-11:
@ne0fhyk Are you alive ?

ne0fhyk, 2016-11-15:
@guiseco yes :) . I've just been very busy for the past couple months with getting [the YouTube VR app out](http://www.theverge.com/2016/11/10/13581046/youtube-vr-review-google-daydream-headest-best-app).
I'll review the pr that's out to fix the issue.

ne0fhyk, 2016-11-15:
Closing this issue at it's being addressed in https://github.com/dronekit/dronekit-android/issues/462 by @billbonney.

closed 2016-11-15 by ne0fhyk