DroidPlanner/Tower #1833: how to run this project in AndroidStudio?

luyuan11233, 2016-11-24:


Linjieqiang, 2016-11-28:
First,you must have Android Studio.And this issue was board and no meaning.

luyuan11233, 2016-11-28:
yes，i have it. but when i run it,it show me this question,
[image]
what to do next? i need your help.
thank you!

luyuan11233, 2016-11-28:
and when i run it,it show follows
Error:(7, 0) Could not find method compile() for arguments [com.google.android.gms:play-services-location:9.4.0] on DefaultExternalModuleDependency{group='com.google.android.gms', name='play-services-maps', version='9.4.0', configuration='default'} of type org.gradle.api.internal.artifacts.dependencies.DefaultExternalModuleDependency.
<a href="openFile:/Users/ruiyunitc/Downloads/Tower-develop/Tower-develop/Android/build.gradle">Open File</a>

is it outdated？so what should i do?

heldib, 2016-12-16:
I am new here too and I am facing the same error above. Anyone can help?
"Error:(7, 0) Could not find method compile() for arguments [com.google.android.gms:play-services-location:9.4.0] ....

billbonney, 2016-12-16:
You need to pick a build variant that you don't need to sign, i.e. devDebug. You open the window by clicking the Build Variant tab of the lower right-hand side
[image]

heldib, 2016-12-16:
This fixed it for me. Thanks a lot!

billbonney, 2016-12-16:
This Issue is closed and kept here for reference. Support is handled here for future http://discuss.ardupilot.org/c/ground-control-software/tower

We try and keep GitHub for issues that require a fix in the code. 👍 Thx

closed 2016-12-16 by billbonney