DroidPlanner/Tower #1836: App Crashes When Connecting to Solo - On Kindle Fire HD7 5th Gen

jasonhamilton, 2016-11-30:
I have a Kindle Fire HD7 (5th Gen) and downloaded Tower from the Google Play Store this week.  All features seem to work in the app, however when I try to connect to my 3DR Solo, a notification says "Unable to set text to speech language" and then the app crashes and exits out.

I've tried about 15 times to connect and successfully connected 1x and cannot get it to connect again.  I've looked at the code and it looks like it allows the connection to still occur even if the TTS error shows, so I'm thinking it's something else causing the problem.

I'm new to Android and this is my only Android device.  Any help on debugging (how to access logs, etc) this would be helpful.

chaviw, 2016-11-30:
Do you have the crash logs? 

Most likely the issue is missing google play services. It relies on that for maps, location, and a bunch of other google specific apis. It's likely to crash because it can't find the classes to interact with play services.

This app was not intended for Kindle devices I don't think ever tested.

There are ways to add support for Kindle, but it would require development work to ensure it uses Amazon specific libraries instead of Google

jasonhamilton, 2016-11-30:
If you point me towards how I would obtain the crash logs, I can post them.

jasonhamilton, 2016-12-01:
Here is a copy of the error log output http://pastebin.com/SP7kCfD4.  The error on line 1 related to DynamiteModule occurs when the program is opened.  The remaining errors occur after I hit the connect link.

jasonhamilton, 2016-12-01:
This appears to be an error with dronekit-android.  I have filed an issue on their repo.  Can someone please confirm?

chaviw, 2016-12-05:
As you can see from the crash logs, it's very clear that it's missing a class specific to google play services. com.google.android.gms is Google Messaging Service so Kindle devices will not have that.

Like I said earlier, if you want to get it to work on Kindle, it will require some work to integrate the compatible libraries from Amazon and use those classes instead of the Google ones when on Amazon device

```
E/DynamiteModule(15009): Failed to load DynamiteLoader: java.lang.ClassNotFoundException: Didn't find class "com.google.android.gms.dynamite.DynamiteModule$DynamiteLoaderClassLoader" on path: DexPathList[[zip file "/data/app/org.droidplanner.android-1/base.apk", zip file "/system/priv-app/MetricsApi/MetricsApi.apk", zip file "/system/priv-app/com.amazon.dp.logger/com.amazon.dp.logger.apk"],nativeLibraryDirectories=[/data/app/org.droidplanner.android-1/lib/arm, /vendor/lib, /system/lib]]
```

chaviw, 2016-12-05:
There does appear to be an issue with the video decoder as well. I'd have to dig into a bit more because `IllegalStateException` is pretty generic and can occur from several ways

ne0fhyk, 2016-12-13:
As pointed by @chaviw, Amazon devices are not supported since they lack Google Play Services.

closed 2016-12-13 by ne0fhyk

billbonney, 2016-12-13:
I think the solution here is to raise an enhancement issue to support Amazon devices. IT will probably be on the backlog until an owner is found to implement it. It's not impossible, but requires some work