DroidPlanner/Tower #1843: Weird thing on my 2 quadcopters

polleke1949, 2017-01-29:
Weird issue on my 2 "old" quadcopters.
It's been a while since I flew these "old" guys, so I was thinking to myself: "Come on, lets get these old birds in the sky again."

I have one with a Crius AIO V1.0 (fw Arducopter 3.1.5-R2) and the other one with a APM mini 3.1 (fw Arducopter V3.2.1)
Both have a BlueTooth module HC-06 on board.
When I connect the Cruis to the Tower app, it won't arm using Tower (with my TX it does) but I CAN refresh the parameters!
When I connect the Mini APM to the Tower App, it won't arm in Tower (with my TX it does) AND it won't refresh the parameters. 
I've tried to re-install the latest firmware on both quadcopters, but no avail.
Tried both Towers (old and Beta) versions, also no avail.

Now for the weird part..... everything used to work like a charm.
What am I doing wrong?

polleke1949, 2017-02-04:
Anybody, any idea?

squilter, 2017-02-05:
The mavlink protocol has changed a lot. I would not count on an up-to-date version of Tower working properly with Copter 3.1.5. You said that you updated the firmware on both copters. Can you verify what version those are running now?

polleke1949, 2017-02-06:
Thank you for your answer. Finaly someone does ;-)
These are my firmwares:
Crius AIO V1.0 (fw Arducopter 3.1.5-R2)
APM mini 3.1 (fw Arducopter V3.2.1)
BTW, I never changed any settings.