DroidPlanner/Tower #1849: Request: Terrain-Following Management in Guided Mode

cbouju, 2017-03-13:
ArduCopter now allow Terrain-Following in Guided Mode, it would be interesting to be able to activate this function with Tower (for Guided and FollowMe modes).

For example, currently this function can be activated with Mission Planner for each WP of a mission by choosing 'Terrain' from the drop-down list of WPs.

Thanks,

billbonney, 2017-03-19:
@cbouju I have this working in a personal branch, I just haven't had time to merge this into the next beta... so coming soon as I find sometime

cbouju, 2017-03-19:
Thanks for your reply, I will watch closely the futures betas.
Must I close the request ?

billbonney, 2017-03-20:
I've reopened it 🙂

Cyberpilot360, 2017-09-04:
Terrain following is a must for precise mapping or agriculture needs! Do we have any news on the progress?