DroidPlanner/Tower #1877: Rotating copter +/- 90 degress in flight direction during waypoint mission

dl2ecc, 2017-11-02:
Hi all,
if I'm flying in a waypoint missing, it would be nice to rotate the copter in a +/- 90 degree angle
to the flight direction. The copter should stay in this angle during the whole missing.
This allows to fly over solar parks while taking infrared videos from solar panels.
This would be a nice feature for the next tower release.

Regards
Wolfgang

gmorph, 2017-11-02:
This isn't a current issue with the code and is really a feature request so shouldn't be in the issues list but on the discussion forum instead.
Further ArduPilot already does this.  In your mission you need to specify a CONDITION_YAW command.  See the documentation here.
http://ardupilot.org/copter/docs/mission-command-list.html#mission-command-list-condition-yaw
Thanks, Grant.

billbonney, 2017-11-02:
You can also set region of interest points along the route and copter will point at those.

closed 2017-11-02 by billbonney

dl2ecc, 2017-11-04:
Hi Grant,

 

thanks for your answer.

I didn't know this function.

I'm going to test it in my next mission.

 

Best Regards

Wolfgang

 

Gesendet: Donnerstag, 02. November 2017 um 23:53 Uhr

http://ardupilot.org/copter/docs/mission-command-list.html#mission-command-list-condition-yaw

dl2ecc, 2017-11-04:
Hi Bill,

 

thanks for your answer.

I try it in my next misiin.

 

Best Regards

Wolfgang

 

Gesendet: Freitag, 03. November 2017 um 00:01 Uhr