DroidPlanner/Tower #1885: Follow Mode Not Working (Pixel Gen1)

funguypro, 2018-02-28:
Hello, I have a 1st gen google pixel and the follow mode is not initiating. The button turns orange and it switches to guided mode. I can adjust the height. It does not set a ROI and it does not follow. I made sure I traveled a fare distance (2000ft) to make sure that it wasn't a distance init problem and it still didn't work. I have tried on 3 google pixel phones and none work with it (even the beta app). My laptops mission planner work fine and the auto mode works fine but follow mode will not work.

I hope this can be solved quickly!

lordneeko, 2018-04-13:
Doesn't work with my Nexus 5 or NExus 7 (2013) either.

lordneeko, 2018-04-30:
with my 3DR Solo. To add information, even though the Tower app says it is in “Guided” mode, my Solo’s transmitter screen says it is in “Brake” mode, which is certainly the behavior that I’m observing when it is up there. Did the Mode enumerations change in Tower or AP? I know the Solo is operating off a pretty old version of AP, but I couldn’t imagine that AP would re-enumerate the modes and break old builds.

lordneeko, 2018-04-30:
I found that the Solex app creater had to make a change to his app as well because apparently the Android GPS accuracy has declined. Might have been an API change...or a constellation issue, but it is consistently not as accurate as it use to be. I cannot find in the Tower app code where the hdop is checked before telling it to follow the target. Is anyone still maintaining this project?

Here is his post.
"Follow/Me Fix:
So, I started getting frequent reports of Follow/Me not working. Follow/Me, unlike all other Follow modes on Solex, uses the device's internal GPS to send locations to the vehicle. To avoid doing a terrible job of it, it filters out locations that have an accuracy of >= 10 meters (higher is worse). If a given location has a higher accuracy value than 10, it doesn't send it to the vehicle. This code has been in place for almost 2 years, with no issues at all. When I started hearing about it not working, I investigated and saw that all of the locations logged in the debug log had an accuracy value of 10 meters at minimum. Not sure what's going on there. Locations have historically had an accuracy value of <= 5 meters on a good Android GPS device. In any case, the 10-meter accuracy meant that no locations were getting sent to the vehicle. Adding to the confusion was the fact that on Simple Follow (the one on the map), I'm actually doing path prediction in those locations to make the path smoother, so I just set the accuracy value for those computed locations to 1 meter (Solo's internal path prediction takes care of it in Follow/Me). So in an effort to make it clearer why it's not working, Solex now displays an error message when the Android location value is inaccurate. To make it actually work, I adjusted the max-accuracy value to 15 meters, and now it works. Strange."