DroidPlanner/Tower #1886: Mapbox Custom Map arent wörking

maxnae, 2018-03-29:
Hello Mapbox has changed custom maps do not have map id anymore.
There is only.
Style URL:
mapbox://styles/maleae/cjf7zonlb1876308597bmkoa
Access token:
pk.eyJ1IjoibWF4bmFl*************************bXhrMjV6MmQifQ.
Could you please fix the error

bikkser, 2019-07-08:
Same issue here! The Tower App requires an ID even though Mapbox only works with Access Tokens

maxnae, 2019-07-09:
Hello have meanwhile found a solution for the problem. If you create a map with Mapbox Classic and upload it, you can use it in Tower.

OlavAa, 2019-08-02:
I actually work to fix this. Going to use the Tower app for a boat and have depth maps integrated into a Mapbox style. There are several problems with this tower version and maps:
(1) Tower can display custom maps from Mapbox in the editor view, but switches over to Google Maps normal view when you select flight data view in Tower. The same issue applies for ArcGIS. And also for offline maps. Bug at least behaves consequent. This part remains to be fixed - work in progress. 
(2) Since mapbox has changed, Tower reports bad credentials for Mapbox. I have decided to modify Tower to use another API (v1 - download styled map tiles). To support this, I've added an input for the user to enter his mapbox user ID. In the mapId field, the user should now enter the ID of the selected style. And the access token still to be filled out, of course. This part is done and is working. But is limited by the first issue. 

When I have completed the changes I can make a fork and upload the resulting code before I modify the app further. This will then basically be Tower v4 with the changes mentioned above, fixing the issue reported. And also fixing a memory leak issue. 
Tower is sadly a dead app, but for me it has all the functionality I need (and then some). Just need to fix the mapbox issue. Just need to figure out why Tower does not load the mapbox maps in "flight data" when it does so successfully in "edit mission"...

OlavAa, 2019-08-10:
Solved (mostly) in develop branch on: github.com/OlavAa/Tower
User must add three attributes in mapbox preferences:
Access token (as before), mapId (but use the STYLE ID here) and the Mapbox username

Before, the mapbox integration was based on Mapbox Project (discontinued) and/or use of the Mapbox Classic app (phased out). This is changed here: Solution relies instead on Mapbox Studio (current). Create a style and customize as desired (import tilesets etc). Then copy the style ID and use that in Tower.

This source has fully functional mapbox maps online/offline in all screens (Flight/Editor/History), but with one flaw;
Tower always (1) load Google Maps (normal) first, then (2) overlays mapbox tiles. #2 always fails when user enters "FLIGHT". But there is a workaround: Change device orientation (eg. portrait/landscape): #2 is run again and the mapbox maps stays visible in the flight screen.

billbonney, 2019-08-10:
Great work, create PR is you like, would be good to keep track of your fix

OlavAa, 2019-08-13:
PR created

maxnae, 2019-08-27:
Hello OlavAa, I have tried to create an apk, but in vain. Could you send me a link to your tower version as Apk?

OlavAa, 2019-08-27:
Some trouble in getting the build environment set up - for sure. I had to resort to an older NDK and said no to a suggested upgrade of grade. And you need to add your Api key for Google Maps to utilize maps. 
MY build env has progressed somewhat (small adaptations for rover, more intuitive utilization of guided mode for carp anglers with bait boats). And I also need to fix the workaround. I can build apk (devDebug)when I have fixed that, for sure.

maxnae, 2019-08-28:
Hello, I also use tower for carp fishing. What I miss here is a possibility to set the marker without a function to save already found spots permanently.  Looking forward to a working tower version of you.

OlavAa, 2019-08-28:
I think you could contribute :) Tower's simplicity makes it a great platform for bait boat use. We have a lot of ideas on how improve Tower besides depth maps, work in progress (nice first step to utilize guided mode already in place, as an example) 
Contact me directly on Olav.aamaas@outlook.com.

OlavAa, 2019-09-12:
> Great work, create PR is you like, would be good to keep track of your fix

@billbonney : Do you have any idea where to look to nail down (to fix the problem) why Tower never displays the mapbox tiles when the flight data view is loaded? This view does support the mapbox tiles: If screen orientation is altered (portrait<=>landscape) the tiles are made visible. But each time the user has used flight history or the mission editor (both views supporting the mapbox tiles flawlessly) and reverts back to flight data, the tiles are gone again (and screen needs to be flipped once again to bring them back)... Looking at the profiler, network activity is almost zero when flight data is called, while editor view or flipping orientation in flight data view causes a spike in network activity (ie loads the tiles).