DroidPlanner/Tower #1897: Google maps not getting displayed on the devDebug Tower app

ansuman87, 2019-03-02:
I am using AndroidStudio for app-development. This is my problem:
After building the app successfully, by choosing devDebug build variant, I have been trying ,albeit unsuccessfully, to make the google maps get displayed on the app. I created a new google map API key, restricted it by providing the android project package name and the SHA1 fingerprint. For the latter, I used the command "keytool -list -v -keystore ~/.android/debug.keystore -alias androiddebugkey -storepass android -keypass android". I recompiled and run the app. But I am still not able to see the map on the app. Where could I have messed up? Though the build was successful, in the androidmanifest file, I got an error in the 'intent-filter' part of the activity 'activities.EditorActivity'. It showed --
missing URL: 
"Ensure the URL is supported by your app, to get installs and traffic to your app from Google Search.  Issue id: AppLinkUrlError." 
I don't even know if the aforementioned was the main issue. I removed the 'intent-filter', as I just wanted to check if the maps are working fine, and tried to run again. But this time also the maps would not get displayed. Any clue is welcome. If more information is needed please let me know.

Thanking you!

billbonney, 2019-03-03:
Try removing the restrictions from you map key. It will work without them. It just limited to the number of requests.

ansuman87, 2019-03-04:
Thanks a lot. I just realized that one has to enable the API key to make the app work. Now it is working fine. Just took a lot of time to figure this out. Meanwhile, I learnt more about Intents. :)

closed 2019-03-04 by ansuman87