DroidPlanner/Tower #1898: Tower smart app discontinued?

souzaux, 2019-04-18:
Personally, was the Droidplanner / Tower smart app discontinued?

AlexShafir, 2019-05-14:
Same question here

AlexShafir, 2019-05-15:
To be honest I would become maintainer itself.

If so I wonder from what to start:

Fork project and nominate myself.
But how to gain attention so that I am not alone in my pursuit?
Obtain write access from main maintainers?
But I doubt it it possible as Fredia (former dev) has left
It seems like fork option is the most suitable

billbonney, 2019-05-15:
I have done some changes to make Tower work with newer releases, I just haven't had the time to make release to the store. I have everything required to do so. Solex is a non-starter as an Open Source project and not pushed changes made to the android dronekit common repo.

You can fork tower or we can figure out how to get changes you have made to the project published as Tower

AlexShafir, 2019-05-15:
@billbonney Great to hear from maintainers!

Alright, I made a fork, where I will lead organized development of Tower.
It is here: https://github.com/AlexShafir/Tower

For now, I see #1 task is to move Tower to MVP architecture

HyGy, 2019-05-15:
That is really cool!!!!

You can give a now name too like: skyscraper :)

HyGy, 2019-05-15:
@AlexShafir the beta tesgint thing is real (at the bottom of the page)?

AlexShafir, 2019-05-15:
> @AlexShafir the beta tesgint thing is real (at the bottom of the page)?

No, I just forked it, that's it

AlexShafir, 2019-05-16:
I made Tower compile with latest Android Studio, Gradle, Kotlin
See: https://github.com/AlexShafir/Tower/tree/resurrection

guiseco, 2019-06-20:
Very Nice!

Surya-prathamzzz, 2025-05-13:
@snorkel123 I'm unable to access the link