DroidPlanner/Tower #1905: Request for removal of dependency of google play services in Tower 4.0010 apk

VU2ATX, 2020-08-29:
Hi,
Can somebody pls remove dependency of google play services in the latest released Tower 4.0010 and post the apk which does not require google play services installed on the device ?

I have a 10'' nice china made Tablet running Tower apk but this tablet does not have google play services. I even tried downloading google play services and installing but was not successful.

Thanks in advance

eagleanurag, 2020-08-30:
Hi! Thanks for your interest. but this version requires the play services to run and it is hard to do what is your requirement for now at the movement. Meanwhile, you can use 3rd party website t download play services for your device please do to you about phone section in your Chinese tablet and fro there get to know about your device distributions, if the x86 app is needed for your device please install play apk of x86 package from that website.

you can refer this blog for more info:

https://www.burrosabio.net/google-play-on-a-chinese-smartphone/

VU2ATX, 2020-09-02:
> Hi! Thanks for your interest. but this version requires the play services to run and it is hard to do what is your requirement for now at the movement. Meanwhile, you can use 3rd party website t [...]

Thanks for your guidance

closed 2020-09-02 by VU2ATX