DroidPlanner/Tower #1910: unknown flight mode Tower app

vmv007, 2021-08-09:
When i connect my copter, Flight modes switch spontaneously from Stabilze to unknown flight mode............ 
I am using pixhawk cube orange arducopter 4 firmware. Pls help me with this issue

RuffaloLavoisier, 2021-11-10:
Which `.APK` do you use for gcs app?

vmv007, 2022-01-23:
> Which `.APK` do you use for gcs app?

[image]

RuffaloLavoisier, 2022-01-23:
> > `.APK`gcs 앱은 무엇을 사용합니까 ?
> 
> [image]

https://github.com/DroidPlanner/Tower/releases/download/Tower-v4.0.0.10/tower-prod-release.400010.apk

Try using this. !

vmv007, 2022-01-23:
> @vmv007
> 
> https://github.com/DroidPlanner/Tower/releases/download/Tower-v4.0.0.10/tower-prod-release.400010.apk
> 
> Try using this. !

Ok

RuffaloLavoisier, 2022-01-23:
@vmv007 
Or [this is](https://cdn.down-apk.com/org.aerohawk.android/Mission%20Maker%20for%20ArduPilot_4.0.0_apkcombo.com.apk?ecp=b3JnLmFlcm9oYXdrLmFuZHJvaWQvNC4wLjAvNC45ZWQ2NjI1ZmE4MTEyYTdiN2ZmZWFmMjdmODU0ZTIxYWJmYmM3OGJhLmFwaw==&iat=1642953266&sig=f1590cda53299293429b6408f2dd7ef6&size=19911150&from=cf&version=latest)

closed 2022-01-24 by vmv007