DroidPlanner/Tower #1912: open in android studio

Zxr1230, 2023-08-22:
I wonder if I am able to open this application as a project in Android Studio and run it on the virtual device in Android Studio. Thank you!

Surya-prathamzzz, 2024-02-20:
the gradle version which is used in this project which is mentioned in the gradle-wrapper.properties file 2.14.1 is too low...
are you able to run this project or not yet ?!

Surya-prathamzzz, 2025-04-28:
@prachideokate7 Sure
how can I help you

prachideokate7, 2025-04-28:
I'm trying to run same repo and It's building up project successfully but not running on emulator!
Can you please tell me How I can do it?

Surya-prathamzzz, 2025-04-29:
@prachideokate7 
I'll delete this soon 
It'll be better to collab just pin me here 8080044997
call or text whatever